open-sdr/openwifi #360: about CSI rader

petrichorZSY, 2023-11-05:
I encountered an error when compiling the second step of CSI rader.     Ubuntu22.04   Vivado2021.1

the second step：
export XILINX_DIR=your_Xilinx_install_directory
(Example: export XILINX_DIR=/opt/Xilinx. The Xilinx directory should include sth like: Downloads, Vitis, etc.)
export OPENWIFI_HW_IMG_DIR=your_openwifi-hw-img_directory
(The directory where you get the open-sdr/openwifi-hw-img repo via git clone)
export BOARD_NAME=your_board_name
(Check the BOARD_NAME definitions in README)

cd openwifi/user_space
./drv_and_fpga_package_gen.sh $OPENWIFI_HW_IMG_DIR $XILINX_DIR $BOARD_NAME
scp drv_and_fpga.tar.gz root@192.168.10.122:openwifi/
scp ./side_ch_ctl_src/side_ch_ctl.c root@192.168.10.122:openwifi/
scp ./inject_80211/* root@192.168.10.122:openwifi/inject_80211/

the error:
/home/zsy/openwifi/adi-linux/tools/objtool/.fixdep.o.cmd:1: *** 缺失分隔符。 停止。
make[4]: *** [Makefile:43：/home/zsy/openwifi/adi-linux/tools/objtool/fixdep-in.o] 错误 2
make[3]: *** [/home/zsy/openwifi/adi-linux/tools/build/Makefile.include:4：fixdep] 错误 2
make[2]: *** [Makefile:64：/home/zsy/openwifi/adi-linux/tools/objtool/libsubcmd.a] 错误 2
make[1]: *** [Makefile:62：objtool] 错误 2
make: *** [Makefile:1628：tools/objtool] 错误 2

How to solve this, please

JiaoXianjun, 2023-11-06:
Could you give a full log that shows what you have input line by line starting from git clone?

petrichorZSY, 2023-11-14:
> Could you give a full log that shows what you have input line by line starting from git clone?

For now, we skipped the second step:
./drv_and_fpga_package_gen.sh $OPENWIFI_HW_IMG_DIR $XILINX_DIR $BOARD_NAME

We find the required file according to the path and copy it to the board:
scp drv_and_fpga.tar.gz root@192.168.10.122:openwifi/
scp ./side_ch_ctl_src/side_ch_ctl.c root@192.168.10.122:openwifi/
scp ./inject_80211/* root@192.168.10.122:openwifi/inject_80211/

Then we follow the quick start in the readme on the openwifi homepage, turn on WiFi and let sdr0 start working, and then execute:
./monitor_ch.sh sdr0 1
and the following steps, and then it can run normally.

Will this cause any problems?

JiaoXianjun, 2023-11-16:
Good to know that you can run the demo.