open-sdr/openwifi #361: Modification of the number of antennas

wangwaa, 2023-11-07:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

wangwaa, 2023-11-07:
Hello teachers:  I'm a graduate student. I would like to ask how to change the antenna's one-receiver to one-receiver-two-receiver or two-receiver-two-receiver? The boards used are zedboard and ad9361s2.   Thank you!

JiaoXianjun, 2023-11-11:
there are 2 tx and 2 rx.
For 2 rx: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/iq_2ant.md
For 2 tx: https://github.com/open-sdr/openwifi/tree/master/doc tx_intf reg_idx 16:

<html>
<body>

16 | tx antenna selection and cdd control | bit1: 0 or 1 to select ant0 or 1. bit4: 1 to enable simple cdd (two antennas have 1 sample tx delay)
-- | -- | --

</body>
</html>

closed 2023-11-16 by JiaoXianjun