open-sdr/openwifi #362: IQ capture cannot receive data

Aniwer, 2023-11-14:
Hardware：Xilinx zedboard + FMCOMMS3
Steps：
1. The board and PC are in the same LAN. The board IP is 10.10.88.122 and the PC IP is 10.10.88.157. Ping is OK.
1. Modify Server address in side_ch_ctl.c `server.sin_addr.s_addr = inet_addr("192.168.10.1")` to `server.sin_addr.s_addr = inet_addr("10.10.88.157")`，and recompile 
2. Follow the quick start command in iq.md，except `insmod side_ch.ko iq_len_init=4095`
3. `loop xxx side count info xxx` normally display
4. Modify the UDP _IP in the python script to 10.10.88.157 
Results：
sock.recvfrom stuck，No txt saving and picture display
Is there something wrong with my configuration? Does the loop display normal mean that the board is sending data normally, but the PC is not receiving it normally?

JiaoXianjun, 2023-11-14:
Do you have issue with all IPs unchanged?

Aniwer, 2023-11-16:
It’s my fault that I didn’t check the startup of the UDP port.

closed 2023-11-16 by Aniwer

chenmo234, 2024-05-22:
Hello, sorry to bother you. May I ask you some questions regarding IQ capture in OpenWiFi? Here is my contact information: c2385405729.