open-sdr/openwifi #363: IQ sample capture - Side info count not incrementing

remydemy, 2023-11-15:
Hello,
I'm trying to execute the IQ sample capture on my ANTSDR E310, but unfortunately the side info count is not increasing.
These are the commands I execute:
```
cd openwifi
./wgd.sh
./monitor_ch.sh sdr0 11
insmod side_ch.ko iq_len_init=8187
./sdrctl dev sdr0 set reg xpu 1 1 // This command is mentioned in iq.md to capture self TX signal, but I'm not quite sure when I should execute it
./side_ch_ctl g
```

The result I then get is: 
```
loop 64 side info count 0
...
```
So the side info count remains at 0 and does not increase.
What could be the cause of that?

Best regards,
Emily

JiaoXianjun, 2023-11-16:
If you read our app note carefully, you should realize that you should not use 8187.

chenmo234, 2024-05-22:
Hello, sorry to bother you. May I ask you some questions regarding IQ capture in OpenWiFi? Here is my contact information: c2385405729.