open-sdr/openwifi #365: openwifi use TDD mode

wangdan25, 2023-11-16:
Can openwifi use TDD mode?Because the SIFS time between each packet in the wifi protocol is 16us, and the VCO needs to be turned back on when switching between states in TDD mode, the synthesizer needs 20us time to achieve lock even in fast lock mode, which is not enough time relative to SIFS, so how to solve this problem?

JiaoXianjun, 2023-11-16:
https://www.sciencedirect.com/science/article/abs/pii/S0140366423002244?via%3Dihub
https://arxiv.org/abs/2204.07354

wangdan25, 2023-11-23:
openwifi reduces TX LO leakage by turning off TX LO in the receiving state, which also reduces turnaround time. However, VCO and DAC are always active, which increases the power consumption of the system.I am curious, is there any rf product on the market that can make the turnaround time meet the requirements of SIFS even if all paths are closed?

JiaoXianjun, 2023-11-23:
In the paper I shared, there are some info (if I remember correctly).