open-sdr/openwifi #366: Cannot boot ZCU102+ FMCOMMS2: [ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004 ]

gnocil, 2023-11-18:
Dear Dr. Jiao, 

I built a bootable SD card for ZCU102+FMCOMMS2 using the latest image openwifi-1.4.0-notter.img.xz following the quick start instructions and I am able to see printouts from UART. However, the boot process stopped in the middle so I cannot get the OpenWiFi system running at the moment. 

Please see the attached log from the UART. The booting ends with an error of [ end Kernel panic - not syncing] and I am unsure if it is caused by the bootable SD card files or something else as I didn't see any similar existing issues from the issue history.

Thank you for looking into the issue! 

```
Xilinx Zynq MP First Stage Boot Loader
Release 2021.1   Mar 14 2023  -  14:21:41
NOTICE:  BL31: v2.4(release):xilinx-v2020.2-2024-g0a69763
NOTICE:  BL31: Built : 07:48:38, Sep 23 2021
PMUFW:  v1.1

U-Boot 2018.01-21439-gd244ce5 (Jul 29 2021 - 16:37:20 +0100) Xilinx ZynqMP ZCU102 revA, Build: jenkins-development-build_uboot-1

I2C:   ready
DRAM:  4 GiB
EL Level:       EL2
Chip ID:        zu9eg
MMC:   sdhci@ff170000: 0 (SD)
*** Warning - bad CRC, using default environment

In:    serial@ff000000
Out:   serial@ff000000
Err:   serial@ff000000
Bootmode: LVL_SHFT_SD_MODE1
Net:   ZYNQ GEM: ff0e0000, phyaddr 15, interface rgmii-id

Warning: ethernet@ff0e0000 using MAC address from ROM
eth0: ethernet@ff0e0000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Device: sdhci@ff170000
Manufacturer ID: 3
OEM: 5344
Name: SL16G
Tran Speed: 50000000
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 14.8 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
reading uEnv.txt
407 bytes read in 19 ms (20.5 KiB/s)
Loaded environment from uEnv.txt
Importing environment from SD ...
Running uenvcmd ...
Copying Linux from SD to RAM...
** No boot file defined **
reading system.dtb
60368 bytes read in 28 ms (2.1 MiB/s)
reading Image
32997888 bytes read in 2195 ms (14.3 MiB/s)
## Flattened Device Tree blob at 04000000
   Booting using the fdt blob at 0x4000000
   Loading Device Tree to 000000000ffee000, end 000000000ffffbcf ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.0-98248-g1bbe32fa5182-dirty (jxj@jxj-pc) (aarch64-xilinx-linux-gcc.real (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35.0.20200730) #4 SMP Tue Mar 14 12:25:14 CET 2023
[    0.000000] Machine model: ZynqMP ZCU102 Rev1.0
[    0.000000] earlycon: cdns0 at MMIO 0x00000000ff000000 (options '115200n8')
[    0.000000] printk: bootconsole [cdns0] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 256 MiB at 0x0000000070000000
[... 390 lines cut ...]
[    6.883742] VFS: Mounted root (ext4 filesystem) on device 179:2.
[    6.900435] devtmpfs: mounted
[    6.904288] Freeing unused kernel memory: 2624K
[    6.908910] Run /sbin/init as init process
Assertion 'clock_gettime(map_clock_id(clock_id), &ts) == 0' faile[    7.396427] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004
[    7.407925] CPU: 0 PID: 1 Comm: init Not tainted 5.10.0-98248-g1bbe32fa5182-dirty #4
[    7.415653] Hardware name: ZynqMP ZCU102 Rev1.0 (DT)
[    7.420601] Call trace:
[    7.423039]  dump_backtrace+0x0/0x190
[    7.426688]  show_stack+0x18/0x30
[    7.429995]  dump_stack+0xd4/0x110
[    7.433389]  panic+0x15c/0x310
[    7.436435]  do_exit+0x8b8/0x904
[    7.439646]  do_group_exit+0x38/0xa0
[    7.443206]  get_signal+0x154/0x76c
[    7.446686]  do_notify_resume+0x228/0x12ec
[    7.450774]  work_pending+0xc/0x39c
[    7.454248] SMP: stopping secondary CPUs
[    7.458164] Kernel Offset: disabled
[    7.461641] CPU features: 0x0040002,20002004
[    7.465892] Memory Limit: none
[    7.468936] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004 ]---
d at src/basic/time-util.c:54, function now(). Aborting.

```

wangdan25, 2023-11-23:
The reason I had this problem earlier was that the SD was locked, you can check SD card

gnocil, 2023-11-23:
> The reason I had this problem earlier was that the SD was locked, you can check SD card

Hi, thanks for sharing your experience. I have checked the lock switch on my SD, but, unfortunately, it is not locked. I have tried to put it on lock mode, and I was having unable to mount error and kernel panic error.

JiaoXianjun, 2023-11-23:
Can you fill in all the info in our template?

Can you share the detailed info of all your operation from downloading the img to inserting the sd card on board?
(The Linux commands and the outputs, the file operations, etc.)

gnocil, 2023-11-23:
> Can you fill in all the info in our template?
> 
> Can you share the detailed info of all your operation from downloading the img to inserting the sd card on board? (The Linux commands and the outputs, the file operations, etc.)

Hi Dr Jiao, 

Please see the updated post on the top with all requested info supplied. 

Basically, I just flashed the .img into a 16GB SD card and have not done anything else. The UART prints Kernel panic errors, so haven't had the chance to input Linux commands yet. I also tried the SD card coming along with the FMCOMNNS2 board, and I can see the linux successfully running with **root@analog:** prompt.

OGRLEAF, 2024-10-29:
I encountered the same issue. I suspect that the RTC is in an abnormal state. During initialization, Kuiper requests the clock, but it fails. The most apparent solution is to set CONFIG_RTC_DRV_ZYNQMP=n in the kernel configuration and then rebuild the kernel. However, please note that the RTC will not function, and the hwclock command will time out.