open-sdr/openwifi #368: spi.v

winfelin, 2023-11-20:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

winfelin, 2023-11-20:
In xpu module, the spi.v can switch the signal from FPGA or ARM. But how to synchronize two clock signals? Or does switching the SPI clock have no impact on AD?
In addition, if the FPGA is transmitting SPI data, that is, the always state machine is active, and the ARM also wants to send SPI data, will it cause its data to not be transmitted correctly?

JiaoXianjun, 2023-11-20:
https://www.sciencedirect.com/science/article/abs/pii/S0140366423002244?via%3Dihub
https://arxiv.org/abs/2204.07354

chshux, 2023-11-20:
spi.v中，FPGA发送SPI数据只控制了RF 板子的TX，当RF 打开TX时RX是不是也是开启的，这样会对RF的TX产生干扰吗？或者说此时RX与TX都运行在同一个频段上，相互之间会不会产生干扰？

JiaoXianjun, 2023-11-20:
Can you read the paper I shared firstly?

JiaoXianjun, 2023-11-20:
> spi.v中，FPGA发送SPI数据只控制了RF 板子的TX，当RF 打开TX时RX是不是也是开启的，这样会对RF的TX产生干扰吗？或者说此时RX与TX都运行在同一个频段上，相互之间会不会产生干扰？

Yes it is possible. But according to our measurements in the paper, the interference is acceptable.

closed 2023-11-20 by JiaoXianjun

winfelin, 2023-11-20:
Thank you very much for recommending this paper. It solved our doubts about SPI competition between CPU and FPGA. For example, if the CPU wants to occupy the SPI bus, it will output the chip select signal 1us in advance, which is enough for the FPGA to complete the SPI data transmission room. of.