open-sdr/openwifi #369: openwifi use FDD mode

wangdan25, 2023-11-20:
openwifi currently uses FDD mode to realize simultaneous receiving and transmitting, but when AP is working, I find that their receiving and transmitting frequency Settings are the same. Isn't FDD requiring different receiving and transmitting frequencies? Won't interference be caused if the frequencies are the same?

JiaoXianjun, 2023-11-20:
https://www.sciencedirect.com/science/article/abs/pii/S0140366423002244?via%3Dihub
https://arxiv.org/abs/2204.07354

wangdan25, 2023-11-20:
Thank you for your answer. I have read this paper.But at present, openwifi uses the normal FDD mode, and the frequency of receiving and sending is the same, and it seems to work normally, here I do not understand how it is to achieve frequency division duplex

wangdan25, 2023-11-20:
Maybe I misunderstood, because I saw that the device tree parameter "adi,frequency-division-duplex-mode-enable" in the zedboard used by openwifi was enabled

JiaoXianjun, 2023-11-20:
yes. you are right. but that does not matter. we set tx rx frequency to the same in the fdd mode.

wangdan25, 2023-11-20:
I seem to understand some, combined with your paper, although the AD9361 register configuration uses FDD mode, but in the TX state will actually turn off the RX LO, so strictly speaking, it is not received and sent at the same time.