open-sdr/openwifi #389: Actual outcome >0.6 ms delay ping time not as shown as short in openwifi introduction.

2018boyan, 2024-01-25:
1. Could you send email to [xianjun.jiao@ugent.be](mailto:xianjun.jiao@ugent.be) to introduce your self?
**Yes, already sent.**

2. Our image is used directly or you build your own image?
**Your image is used directly. I used the image "openwifi-1.4.0-notter.img" unzipped from "openwifi-1.4.0-notter.img.xz".**

3. What is your own modification?
**No modification**

4. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
**Ubuntu 20.04, Just use openwifi-1.4.0-notter.img for Quick start.**

5. Board/hardware type
**Antsdr E316(AD9361 + Xilinx Zynq 7020 SoC)**

6. WiFi channel number
**Default  5g（36）and 5g （48）**

7. Steps to reproduce the issue, and the related error message, screenshot, etc
[image]
Use a windows computer (win 11) to connect openwifi and ping the ap.
[image]
Use the other e316 (AD9361 + Xilinx Zynq 7020 SoC) with ap-client-two-sdr mode.
[image]
[image]
Change to 36 channel
[image]
[image]
Change to 2.4g channel
[image]
[image]

Use ad-hoc-two-sdr mode
[image]
[image]
[image]
[image]

How to get very low delay ping time as shown in introduction? Should I change some configs? Or change devices? Thanks in advance!

JiaoXianjun, 2024-01-26:
You can try to play with some parameters/registers, like:
https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md
https://github.com/open-sdr/openwifi/blob/master/doc/README.md

2018boyan, 2024-01-30:
> You can try to play with some parameters/registers, like: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md https://github.com/open-sdr/openwifi/blob/master/doc/README.md

I get it! I am all gratitude!