open-sdr/openwifi #394: enable 802.11p

nits-skydeploy, 2024-03-08:
Hi Xianjun,
My name is Nitin, first of all let me congratulate you. Openwifi is really very neat and strong work from you, thanks for doing it and sharing it. 

    1. Our image is used directly or you build your own image?
    I am building image from scratch as er kuiper.md.

    2.What is your own modification?
    I am trying to use 802.11p with 10 MHz bandwidth, i have made changes in verilog files and rf_init_11n.sh, also i designed a filter for 10MHz from MATLAB.

    3.Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
    Focal Fossa, Vivado 2021.1, open wifi 1.4.0

    4. Board/hardware type
    Antsdr_e200

    5. WiFi channel number
    172,176, 178

    6. Steps to reproduce the issue, and the related error message, screenshot, etc
    when running sdr-ad-hoc-up.sh sdr0 5900 192.168.13.1, it gives me channel disabled error, although i modified the sdr.h to include channel 178 for 5900MHz channel, and also modified sdr_regd to include the said channel.

    7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
    not tried

    Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

    Any other thing we need to know for helping you better?
    I just need to understand if I change the fpga design to decimate and interpolate for 10MHz instead of 20MHz, use appropriate filter and do not modify the sdr.h file, can i get the effective bandwidth as 10MHz on any channel (except those that are disabled in sdr.h).

    2nd question is how do I enable 802.11p mode in openwifi?

    My 3rd question is regarding running openwifi on arbitrary frequency, even if i run ./sdrctl dev sdr0 set reg rf 1 3500 and ./sdrctl dev sdr0 set reg rf 5 3500, and then start ad-hoc network using ./sdr-ad-hoc-up.sh sdr0 3500 192.168.13.1 it gives me error "Error for wireless request; set frequency".
    So how do i make sure that my wifi is running on 3500MHz, and how do i connect client on this frequency (suppose i have 2 antsdr_e200)?

    Please look into my questions and if possible provide some hints or solution towards the same.
    Thanking you in anticipation.
    Nitin Kumar

JiaoXianjun, 2024-03-26:
To change the BW/sampling rate, I would suggest you check information in this issue : https://github.com/open-sdr/openwifi/issues/155

nits-skydeploy, 2024-03-27:
Hi Xianjun,
Thanks for your reply. I have tried the same and made modifications in the
design including the filters, verilog files and .sh files but I don't know
how to check whether the changes are working or not. Shall i make two of
such boards and try to make adhoc network with 10MHz bandwidth and then
check for the data rate with iperf? Also can i change the freq to any value
other than 5.9GHz, as 10MHz is allowed over ITS band right now.

Regards
Nitin Kumar

nits-skydeploy, 2024-03-28:
Hi Xianjun,
I tried making two radios with 10MHz bandwidth, i am able to run the script
for ad-hoc network with channel 174 and 10MHz bandwidth, but the two radios
do not connect on same network, the cell ID for both are different. Can you
suggest where i may be wrong, additionally please suggest how the linux
kernel supports 1/2/5/10 MHz bandwidth.

Thanks and regards
Nitin Kumar

Denisfk1985, 2024-04-14:
Hello! Any progress with enabling 802.11p? This topic is also very interesting to me. I was just able to launch openwifi on the LibreSDR platform and now I’m thinking about how to enable the 10 MHz band and solve the problem with frequencies. I was not able to set it above 5900 MHz, only if directly in the registers using ./sdrctl.

nits-skydeploy, 2024-04-15:
hi, I did what was suggested by @JiaoXianjun, but it is not working out. I can't connect any user to the wifi, also not able to connect two devices using ad-hoc network.

JiaoXianjun, 2024-04-20:
Maybe you could search/learn how other chip support 5MHz and 10Mhz in the kernel driver tree, like ath9k (if my memory is correct).

nits-skydeploy, 2024-04-22:
ok @JiaoXianjun I will look into this. Thanks for sharing

brainrecursion, 2026-09-29:
Did anyone work out how to change the bandwidth to 10MHz, there are a few issues asking for this but I haven't seen an answer of how to do  this.