open-sdr/openwifi #396: CSI Phase keeps changing Despite relatively stable environment

ThETyMaNp, 2024-03-18:
Hi, I tried to use this demo and was able to get a better looking constellation(https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/radar-self-csi.md), but the problem is thar the CSI phase keeps changing abruptly despite relatively stable environment.

[What is your own modification?]I modified the code in python so that it displays the csi and its phase, as well as the frequency deviation.

I measured the CSI while keeping the transceiver antenna stationary and the surroundings relatively stable (I even went to a darkroom to measure it), but found that even though the surroundings were relatively stable, the phase of the CSI still changed abruptly, and the value of this abrupt change is relatively regular.I initially thought it was due to a sudden phase change in the CSI due to the frequency offset, however, during later tests, the CSI still had a sudden phase change with a frequency offset of 0 (the amplitude information was normal)
[image]
[image]
I'm a sophomore in college and this question has been bothering me for a long time, so I'm taking the liberty of asking for your advice, and I hope that you'll come to help me out!
Sincerely,
Guang Liu.

JiaoXianjun, 2024-03-19:
to have stable wifi radar signal/phase, you need to use the iq capture functionality:
https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/packet-iq-self-loopback-test.md#iq-self-loopback-quick-start 

Please use the trigger condition 3:
3 	the tx_intf_iq0 becomes non zero (the 1st I/Q out)
https://github.com/open-sdr/openwifi-hw/blob/900aa8f4c330d047660e10ac871fff922f44aa98/ip/side_ch/src/side_ch_control.v#L460C35-L460C55

You should start do this from a setup with tx antenna rx antenna very close, and a quite low rx gain (check how to set rx gain in manual gain mode not AGC gain mode : https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md  )

You can monitor the iq plot via our python script, and adjust the rx gain manually to have IQ amplitude not too big (clipping) not too small (too low SNR).

After you have stable iq loop back over the air (closely co-located tx and rx antenna), you should see very stable stf and ltf in the captured IQ real-time plot.

You can calculate the IQ sample phase difference between "all captures" and "the 1st captures". it is very stable. 

You can also easily write a python script to calculate real-time CSI based on captured ltf IQ. (fft, divided by the tx ltf frequency domain signal, etc.)

ThETyMaNp, 2024-03-19:
Thx very much for your help！I'll try it as soon as I can

ThETyMaNp, 2024-03-20:
Thank you again for your sincere help, I followed your instructions, first I put the transceiver antennas very close to each other (I used two horn antennas about 30cm apart), then I adjusted the gain of the RX to 30 (I got a gain of about 41 at the actual reception), and adjusted the trigger mode to 3. Here are the commands I used to run it:

>In one ssh:
ssh root@192.168.10.122
cd openwifi
./wgd.sh
./monitor_ch.sh sdr0 44		
./sdrctl dev sdr0 set reg xpu 8 1000	
insmod side_ch.ko iq_len_init=4095	
./side_ch_ctl wh11d100		
./side_ch_ctl wh8d3		
./sdrctl dev sdr0 set reg xpu 1 1	
./side_ch_ctl wh5h0	
./set_rx_gain_manual.sh 30		//30dB
./side_ch_ctl g0	

> In another ssh:
cd openwifi/inject_80211/
(make)
./inject_80211 -m n -r 5 -n 1 sdr0

> on computer:
python iq_capture.py 4095

I then ran it several times in a second ssh session:
 /inject_80211 -m n -r 5 -n 1 sdr0

Here are the IQ figures I got in order:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Question:
Since I don't really understand what stf and ltf sequences are supposed to look like or how to tell if they are stable or not, I'll take the liberty of bothering you and ask if this is considered to be obtaining a stable ltf sequence?
If so,can I start the next step of calculating the csi?

I sincerely hope that you can give me some advice.Thank you again!!!：）
Sincerely,
Guangteng Liu.

petrichorZSY, 2024-04-12:
Hello, I would like to ask you how you process CSI, frequency offset and other information, and how you print it out. What does this information have to do with the side_info.txt file? I would really appreciate it if you could reply to me

ThETyMaNp, 2024-04-12:
Hello,

You can obtain the measurement information by running side_info_display.py, which will save it to side_info.txt for further MATLAB data processing. CSI and frequency offset data can also be acquired by creating Python scripts based on side_info_display.py.

JiaoXianjun, 2024-05-03:
You might also need to disable our Tx LO automatic control. Because each time LO ON/OFF, the phase might be different. 

Check the xpu register 13. Set it to 1 will disable it.

ThETyMaNp, 2024-05-03:
Thank you very much for your prompt response. I have made an attempt to incorporate control for register 13 at the indicated location. The code is as follows:
`ssh root@192.168.10.122
cd openwifi
./wgd.sh
./monitor_ch.sh sdr0 44		
./sdrctl dev sdr0 set reg xpu 8 1000	
insmod side_ch.ko iq_len_init=4095	
./side_ch_ctl wh11d100		
./side_ch_ctl wh8d3		
(./side_ch_ctl wh13d1)
./sdrctl dev sdr0 set reg xpu 1 1	
./side_ch_ctl wh5h0	
./set_rx_gain_manual.sh 30	
./side_ch_ctl g0	`
But the results I obtained are still as follows：
[image]

ThETyMaNp, 2024-05-03:
Made it!!!
Thanks for your generous help!!!My gratitude is beyond words!
：）

closed 2024-05-03 by ThETyMaNp