open-sdr/openwifi #397: extracting CSI  results from two RX antennas

wuhhaCos, 2024-04-02:
Dear Professor Jiao,

I am currently working on your OpenWiFi project, primarily focusing on CSI extraction. I have encountered a couple of questions:

1. In the CSI extraction process, does it have to work in the full-duplex (Self-TX) mode? Can we set it as a receiver to obtain CSI?

2. If full-duplex mode is required, does the CSI extraction process only work for single RX sampling? The hardware platform we used is Xilinx Zed board + FMCOMMS3, which includes two Rx antennas.

3. If the above options are not available, can I modify the FPGA code in your project to enable the entire module to operate as a receiver for CSI extraction and sample equalizer results from 2 RX?

I am eager to contribute to the OpenWiFi platform and would greatly appreciate your guidance on these matters.

Please find the corresponding GitHub link for reference: [CSI Extraction in OpenWiFi](https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi.md)

Thank you for your assistance.

chenmo234, 2024-05-22:
Hello, sorry to bother you. May I ask you some questions regarding CSI capture in OpenWiFi? Here is my contact information: c2385405729.

JiaoXianjun, 2024-06-14:
Check: https://github.com/open-sdr/openwifi/issues/411