open-sdr/openwifi #399: Modifying the bandwidth can impact the stability of work

huangfu001206, 2024-04-07:
Hello, Mr. Jiao. We attempted to modify the bandwidth to **2.5 Mbps** by adjusting the corresponding value in the "rf_init_11n.sh" file (dividing the original value of 20M by 8) and generated our own filters. 
The program runs successfully, but it cannot run for an extended period. Typically, after running for 1 hour, the **operating system freezes** completely. We would like to know the possible reasons for this issue. Thank you very much.

Denisfk1985, 2024-04-14:
Hello! Check the status of the viterbi license; it may be in demo mode; in this case, after about an hour of operation, viterbi stops functioning.
Was it difficult to make 2.5 Megabits\s? What is the width of the channel? I want to make 10 MHz, but I still don’t know what to do, where to start.

AnVuNam, 2025-02-06:
Could you please share the steps you did?
I tried reducing the values ​​of **in_voltage_rf_bandwidth** and **out_voltage_rf_bandwidth** in the **rf_init_11n.sh file**, as well as editing the corresponding fir file, but nothing changed.