open-sdr/openwifi #401: capturing CSI for 2 antenas

DusanKvasnicak, 2024-04-16:
Dear Professor Jiao,

I am currently using your OpeneWifi implantation of wi-fi chip for my Thesis.
Topic of my thesis is to use OpenWifi implementation of wi-fi chip and create program that will show distance and movement of devices. I want to use SpotFi algorithm. 

My question is: Is there possibility to capture CSI information from 2 antennas ?
SpotFi paper: Paper:SpotFi: Decimeter Level Localization Using WiF

Thank you for your assistance.

JiaoXianjun, 2024-04-17:
Yes it is possible. 
Currently the advised way is capturing dual antenna I/Q samples, and calculate dual antenna CSI based on LTF.
https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/iq.md
https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/iq_2ant.md

DusanKvasnicak, 2024-04-17:
Than you for your quick response. I have one more question: I's there a possibility to capture RSSI information form both antennas separately?

JiaoXianjun, 2024-04-17:
You can try to calculate dual antenna rssi based on the dual antenna iq sample. (considering you know/set the manual gain of the two ad9361 channels)

DusanKvasnicak, 2024-04-18:
Thanks for your quick response and informations

closed 2024-04-18 by DusanKvasnicak