open-sdr/openwifi #405: rssi correction

wangdan25, 2024-05-13:
Hi xianjun,
I'm confused about how the RSSI value is obtained, and why is the rssi_half_db_offset 150? If I want to use the 1G band, how should I determine these values?
```
if (freq_MHz<2412) {
rssi_correction = 153;
} else if (freq_MHz<=2484) {
rssi_correction = 153;
} else if (freq_MHz<5160) {
rssi_correction = 153;
} else if (freq_MHz<=5240) {
rssi_correction = 145;
} else if (freq_MHz<=5320) {
rssi_correction = 145;
} else {
rssi_correction = 145;
}
```

chenmo234, 2024-05-22:
Hello, sorry to bother you. May I ask you some questions regarding IQ capture in OpenWiFi? Here is my contact information: c2385405729.

JiaoXianjun, 2024-05-22:
> Hi xianjun, I'm confused about how the RSSI value is obtained, and why is the rssi_half_db_offset 150? If I want to use the 1G band, how should I determine these values?
>
> ```
> if (freq_MHz<2412) { [...]

This is kind of "manual calibration" values.
If the RSSI in your new band is not accurate, you need to add the new calibration/correction value inot that piece of code.