open-sdr/openwifi #407: Instability in CSI Data Obtained by OpenWiFi When Monitoring Mobile Phone WiFi Signals in a Stable Environment

LvHaoyuisme, 2024-05-18:
Hello, Dr. Jiao!
We are attempting to receive actual WIFI signals on the OpenWiFi platform, using a single channel plus RIS reconfigurability to simulate multi-channel for direction finding of WIFI devices.

chenmo234, 2024-05-22:
Hello, sorry to bother you. May I ask you some questions regarding IQ capture in OpenWiFi? Here is my contact information: c2385405729.

chenmo234, 2024-05-23:
Can sdr0 continue to be used as an AP while in monitor mode?Does your monitored device need to be tuned to channel 1 to be monitored?

chenmo234, 2024-05-23:
After running ./fosdem.sh, I can detect OpenWiFi with my phone, but after running ./monitor_ch.sh sdr0 11, my phone can no longer find the OpenWiFi AP.

LvHaoyuisme, 2024-05-23:
I've encountered a similar issue where the device seems unable to operate both as an access point and to monitor a specific channel at the same time. After running ./fosdem.sh, you can proceed with IQ sampling or obtain CSI data without needing to run ./monitor_ch.sh sdr0 11 to monitor a specific channel.

chenmo234, 2024-05-23:
Thank you very much for your response! I'll try it right away. I've been stuck on this for two days

chenmo234, 2024-05-23:
It would be an honor to stay in touch with you. May I add your contact information?

JiaoXianjun, 2024-05-30:
> Hello, Dr. Jiao! We are attempting to receive actual WIFI signals on the OpenWiFi platform, using a single channel plus RIS reconfigurability to simulate multi-channel for direction finding of WIFI devices.But we found that under the same conditions, the CSI data varies greatly, especially the phase of the CSI. I saw #396, so I followed the instructions in that issue and obtained stable CSI with the openwifi device operating in a self-transmitting and self-receiving mode. ![屏幕截图 2024-05-18 174829](https://private-user-images.githubusercontent.com/133653103/331790355-943b8010-8724-477b-952d-fc42b14be414.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwODQwNTYsIm5iZiI6MTcxNzA4Mzc1NiwicGF0aCI6Ii8xMzM2NTMxMDMvMzMxNzkwMzU1LTk0M2I4MDEwLTg3MjQtNDc3Yi05NTJkLWZjNDJiMTRiZTQxNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUzMFQxNTQyMzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04NjQ4MDEwZTg0YjgwMWRjNDI5MmI5ODI1OWQ2YzUwMDNlMzdjNjdhOWM4MzJhNDZhMzA2NmJjMjgwYWE2Y2M1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xi_tMrlrTeQUGQufVvK2kZPb4WMoGsC4-jO8Cp3zAS8) The result is perfect. I tried to monitor the WiFi signal from a specific device in monitor mode and obtain the corresponding CSI, but the results were very poor. My operations are as follows:
> 
> > In one ssh:
> 
> > `cd openwifi`
> > `./wgd.sh`
> > `insmod side_ch.ko`
> > `./monitor_ch.sh sdr0 1`
> > #The target device is operating on channel 1
> > `./side_ch_ctl wh1h4001`
> > `./side_ch_ctl wh7h00cf3e83`
> > `./side_ch_ctl wh8d3`
> > `./sdrctl  dev sdr0 set reg xpu 1 1`
> > `./sdrctl  dev sdr0 set reg xpu 13 1`
> > `./side_ch_ctl g0`
> 
> You mentioned setting the receive gain to a fixed value to avoid the impact of AGC ,so I opened another SSH.
> 
> > In anther ssh:
> 
> > ` ./set_rx_gain_manual.sh 10`
> > #I have tried setting different gains and found that when the receive gain is too low, I cannot obtain CSI data. I have tried 10, 20, 30, 40, and 50, but although I can get CSI data, it is very unstable.
> > ![Figure_1](https://private-user-images.githubusercontent.com/133653103/331791716-9fa88f0b-5a4a-4b76-9f31-98338e338c8e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwODQwNTYsIm5iZiI6MTcxNzA4Mzc1NiwicGF0aCI6Ii8xMzM2NTMxMDMvMzMxNzkxNzE2LTlmYTg4ZjBiLTVhNGEtNGI3Ni05ZjMxLTk4MzM4ZTMzOGM4ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUzMFQxNTQyMzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yYjU3ZTU4MzY3N2M5ODg3Yzk2YTM4Nzk5M2IzN2YwYTMzODQ0ODJlMzBjM2ViMWZlYjA1ZGUxYzYyNTlhNzk2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.N-n0RsDSrV_MjOHaUyrSb2jDugE3VlZVPVEIR2AFaDE)
> > I used the test_side_info_file_display.m script to plot the side_info.txt collected from a single acquisition ( I made a small modification to unwrap the phase)
> 
> ![屏幕截图 2024-05-18 183054](https://private-user-images.githubusercontent.com/133653103/331792780-89cc0550-e88b-4d27-95bf-9b1b098e106f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwODQwNTYsIm5iZiI6MTcxNzA4Mzc1NiwicGF0aCI6Ii8xMzM2NTMxMDMvMzMxNzkyNzgwLTg5Y2MwNTUwLWU4OGItNGQyNy05NWJmLTliMWIwOThlMTA2Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUzMFQxNTQyMzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lNDUyNjMwYjM0OWZjYWUxZmQ3ZjQ4YWU0ZTViNGU1ZTVkOTMzNWU2YTFlMzU2OTI4MDgwNTE3NzBhYTkzZmMwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.FdhF3UJ-i0XnNglEdZrGTUT5OP0j_4HdjH0TSz_GTi4)
> 
> The results are very poor.The phase and amplitude of the CSI are both unstable.
> 
> Previously, I referred to https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi.md, and the obtained CSI data amplitude was relatively stable, but the phase remained unstable. My operations:
> 
> > `cd openwifi`
> > `./wgd.sh`
> > `insmod side_ch.ko`
> > `./monitor_ch.sh sdr0 1`
> > `./side_ch_ctl wh1h4001`
> > `./side_ch_ctl wh7hd88d2a2f`
> > `./set_rx_gain_manual.sh 10`
> > `./side_ch_ctl g0`
> > ![屏幕截图 2024-05-18 184201](https://private-user-images.githubusercontent.com/133653103/331793450-f3fdb5fa-8d0e-4ce2-9528-ca965bfd96cd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwODQwNTYsIm5iZiI6MTcxNzA4Mzc1NiwicGF0aCI6Ii8xMzM2NTMxMDMvMzMxNzkzNDUwLWYzZmRiNWZhLThkMGUtNGNlMi05NTI4LWNhOTY1YmZkOTZjZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUzMFQxNTQyMzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNjRhYjk0YWQ1MTE1ZGM2M2RhYjJlNmI3NWJiZGNlZjM1ZDQyYzQ0NDBhZjZmZDkxN2ViZGNhYjkzZTYwZTQ4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.M_mu30urjp6SebhZh7kdu9JZ5L6hU3_8rS1KKxJ07gw)
[... cut ...] JGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUzMFQxNTQyMzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wYWZkOWNhODdjOGMzYjA1YTNlOTMyN2UzYmU2NDZkNTY1MDMxYmM5NzcwYTI2MDI5ZDIwYTYwMmIyMzU2NDk2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ssHNV4ippc8OrvyIfIaE3pyrqkWNh3J9i3li35_Yky4)
> > The device I am using is SDRpi
> > What should I do to obtain stable CSI data from actual WiFi devices?

Please re-format the issue according to our issue template.