open-sdr/openwifi #411: How to extracting CSI results from two RX antennas for the phase difference

Remaker-JDY, 2024-06-05:
1. Our image is used directly or you build your own image?
use yours directly(openwifi-1.4.0-notter.img).

2. What is your own modification?
no modification.

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
none

4. Board/hardware type
zed_fmcs2

5. WiFi channel number
none

6. Steps to reproduce the issue, and the related error message, screenshot, etc
none

8. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
none

10. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods
none

12. Any other thing we need to know for helping you better?
 (1) In the CSI extraction process, does it have to work in the full-duplex (Self-TX) mode? Can we set it as a receiver to obtain CSI?

(2) If full-duplex mode is required, does the CSI extraction process only work for single RX sampling? The hardware platform we used is Xilinx Zed board + FMCOMMS3, which includes two Rx antennas.

(3) If the above options are not available, can I modify the FPGA code in your project to enable the entire module to operate as a receiver for CSI extraction and sample equalizer results from 2 RX?

JiaoXianjun, 2024-06-07:
Please follow our issue template.

JiaoXianjun, 2024-06-12:
You need to do like this:
1. Use dual antenna IQ sample capturing: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/iq_2ant.md
2. calculate the phase difference between the dual antenna IQ (any algorithm you like, or like what he did: https://github.com/open-sdr/openwifi/issues/396)