open-sdr/openwifi #412: Error reporting about CSI radar

csy990406, 2024-06-12:
Hello, Dr Jiao,
I find the following issue while trying to implement CSI radar (https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/radar-self-csi.md). According to the instructions, I need to execute the following command: "./drv_and_fpga_package_gen.sh $OPENWIFI_HW_IMG_DIR $XILINX_DIR $BOARD_NAME". However, an error occurs at line 42 of the openwifi/user_space/drv_and_fpga_package_gen.sh with the following command: unzip $OPENWIFI_HW_IMG_DIR/boards/$BOARD_NAME/sdk/system_top.xsa -d ./hdf_and_bit. I believe the issue is that the openwifi-hw/boards/sdk folder and the system_top.xsa file cannot be found. Where can I go to get /sdk/system_top.xsa?
Thanks for your help.

JiaoXianjun, 2024-06-12:
Please re-format the issue according to our issue template.

csy990406, 2024-06-12:
> Can you share what have you done step by step (command and the outputs) from git clone to get that error?
[image]

csy990406, 2024-06-12:
> Can you share what have you done step by step (command and the outputs) from git clone to get that error?
csy@PC:~$ export XILINX_DIR=/opt/Xilinx
csy@PC:~$ export OPENWIFI_HW_IMG_DIR=/home/csy/openwifi-hw
csy@PC:~$ export BOARD_NAME=antsdr
csy@PC:~$ cd openwifi/user_space
csy@PC:~/openwifi/user_space$ ./drv_and_fpga_package_gen.sh $OPENWIFI_HW_IMG_DIR $XILINX_DIR $BOARD_NAME
$XILINX_DIR is found!
$BOARD_NAME is found!
$OPENWIFI_HW_IMG_DIR is found!
unzip:  cannot find or open /home/csy/openwifi-hw/boards/antsdr/sdk/system_top.xsa, /home/csy/openwifi-hw/boards/antsdr/sdk/system_top.xsa.zip or /home/csy/openwifi-hw/boards/antsdr/sdk/system_top.xsa.ZIP.
$BIT_FILENAME does NOT exist. Please check!

JiaoXianjun, 2024-06-12:
which app note are you following? I remember that we describe the meaning of OPENWIFI_HW_IMG_DIR in some app note. could you look for it.

csy990406, 2024-06-13:
> which app note are you following? I remember that we describe the meaning of OPENWIFI_HW_IMG_DIR in some app note. could you look for it.

Thank you for your response. After re-downloading the openwifi-hw-img, the issue has been resolved.

closed 2024-06-14 by JiaoXianjun