open-sdr/openwifi #414: [Feature request] Support of SoapySDR as a neutral and platform independent SDR support library

ohault, 2024-06-14:
Currently, OpenWifi supports a list of supported hardware (see - https://github.com/open-sdr/openwifi).

This ticket aims to leverage the [SoapySDR API](https://github.com/pothosware/SoapySDR/wiki) as a neutral and platform independent SDR support library.

JiaoXianjun, 2024-06-14:
I checked soapysdr lib. Seems like it is a lib to help do IQ sample streaming between the computer and the antenna.

openwifi is a Wi-Fi chp design (currently it is FPGA still). It works just like any other commercial Wi-Fi devices, one end is antenna, the other end is the Wi-Fi/Ethernet packet.

Could you elabrote more about the feature required?

JiaoXianjun, 2024-06-14:
> It would be in the case of SoftMAC drivers architecture towards SDR hardware.

Openwifi design is already softmac and has Linux driver.

ohault, 2024-06-17:
Can I understand that SoapySDR would only make sense in the case of running at least a part of 802.11PHY on a real-time OS host? 

For [OpenWifi-hw](https://github.com/open-sdr/openwifi-hw) implementation, is there an existing open toolchain/API that could be used to abstract supported FPGA platforms ?

JiaoXianjun, 2024-06-18:
I am not fully understanding the question.
Could you describe a bit further by an example? For instance, use which hardware supported by SoapySDR to achieve what kind of functionality?

JiaoXianjun, 2024-06-20:
> > I am not fully understanding the question.
> > Could you describe a bit further by an example? For instance, use which hardware supported by SoapySDR to achieve what kind of functionality?
> 
> About Q1, I can think about 6 GHz transceiver with a SoapySDR driver to be use then with a project like https://github.com/bastibl/gr-ieee802-11
> 
> About Q2, for this project the question would become how to be more agnostic in regards to FPGAs and not limited to a set of supported target FPGAs

For 6 GHz transceiver with a SoapySDR running gr-ieee802-11, you can use any hardware SoapySDR supported. openwifi is not a SDR for IQ sample streaming. I don't see strong connection in this scenario.

"more agnostic in regards to FPGAs", Even in SoapySDR, to add new hardware support, you need to do some work (like driver) in SoapySDR, right?

In Wi-Fi domain, this layer is the Linux mac80211 framework. Above it, it is hardware agnostic, below it, it is hardware specific driver for different Wi-Fi chip/dongle/devices.

Again, I am not fully sure what do you want. There are already many SDR that can do IQ sample streaming to computer. And SoapySDR is great to offer such functionality.

openwifi is about implementing WiFi NIC in FPGA and offer standard Wi-Fi driver/interface in Linux, not IQ sample streaming.

ohault, 2024-06-20:
About 

> For 6 GHz transceiver with a SoapySDR running gr-ieee802-11, you can use any hardware SoapySDR supported. openwifi is not a SDR for IQ sample streaming. I don't see strong connection in this [...]

About Q1, I agree, it is not directly related to openwifi project. I will create a ticket on https://github.com/bastibl/gr-ieee802-11 about how to rely on SoapySDR in complement to the current support of some USRP devices via UHD ?

About Q2, I will create a new dedicated ticket on this topic because it is not related to SoapySDR.

closed 2024-06-20 by ohault