open-sdr/openwifi #416: MCS problem

jyskr, 2024-06-25:
Hello, Dr. Jiao!
In high-throughput and non-high-throughput modes, you can control the Modulation and Coding Scheme (MCS) using ./sdrctl dev sdr0 0/1 x. In the sdrctl documentation, it is mentioned that when x is set to 0, the MCS is automatically controlled by Linux. I would like to know if this automatic switching of the MCS is based on channel characteristics, and where exactly is the code that handles this?

JiaoXianjun, 2024-06-25:
Please re-format the issue with our template. Thanks.

jyskr, 2024-06-26:
> Please re-format the issue with our template. Thanks.

Dear Dr.Jiao, I have re_formatted it and encountered two issues that I need help with.

HavingaThijs, 2024-07-05:
> If this automatic switching of the MCS is based on channel characteristics?

Linux uses the Minstel algorithm which looks at several metrics, including throughput estimation, see e.g.: https://wireless.wiki.kernel.org/en/developers/documentation/mac80211/ratecontrol/minstrel 

> Where exactly is the code that handles this?

For legacy packets, it would be [here](https://github.com/analogdevicesinc/linux/blob/b6e379910a11af77e6500ed8b0895006e471a279/net/mac80211/rc80211_minstrel.c) and for HT packets [here](https://github.com/analogdevicesinc/linux/blob/b6e379910a11af77e6500ed8b0895006e471a279/net/mac80211/rc80211_minstrel_ht.c).

jyskr, 2024-07-05:
> Linux uses the Minstel algorithm which looks at several metrics, including throughput estimation, see e.g.: [...]

Thank you very much for your response! It will help me a lot.

closed 2024-07-05 by jyskr