open-sdr/openwifi #447: Assistance with Running OFDM Project on ZCU102 + FMCOMMS3 in Vitis

MehmettAkiff, 2025-03-04:
Hello,

I have successfully implemented the Analog Devices reference project using the ZCU102 and FMCOMMS3 platforms. Now, I would like to run your OFDM project as part of my university graduation project. However, I do not have access to the C code for the Vitis environment.

As I do not have experience with Linux-based environments, I would appreciate your guidance in providing the necessary Vitis code or instructions on how to adapt the project to work in the Vitis environment on my hardware setup.

You can reach me via email at 21050211078@aybu.edu.tr for further communication.

Any help, documentation, or reference projects would be greatly appreciated!

Best regards,

A-T-Kristensen, 2025-03-07:
Hello Mehmet,

What exactly do you need to do on the openwifi platform? "Standard" C code programs can be written and compiled on the board, you don't need Vitis for that. 

Regards, Andreas

ViruKM, 2025-04-26:
Hello Kristensen,

I am trying to implement Openwifihw repository 
Trying to build FPGA in vivado for board antsdr_e200 I followed all steps after I gave source ./openwifi.tcl in vivado lot of errors are coming related to IP’s could you please help me out for this