open-sdr/openwifi #448: Change any frequency

guaguaguaguaguayaya, 2025-03-06:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
Yes, already sent.
1. Our image is used directly or you build your own image?
Your image is used directly. I used the image "openwifi-1.4.0-notter.img" unzipped from "openwifi-1.4.0-notter.img.xz".
2. What is your own modification?
No modification.
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
Ubuntu 22.04
4. Board/hardware type
ANTSDR E200/AD9361
5. WiFi channel number
5G（36）/2.4G（6）
6. Steps to reproduce the issue, and the related error message, screenshot, etc
Dear Dr. Jiao, I have the following questions about arbitrary frequency regulation:
（1）The first is that I work fine at both 5GHz and 2.4GHz, and the hostapd-openwifi.conf file shows as follows:

[image]
When the AP is successfully connected with the STA, the AP displays:

[image]
The connected oscilloscope is displayed

[image]
Based on https://github.com/open-sdr/openwifi/blob/master/doc/README.md , I followed the steps below to make the change, and I want to change the frequency to 70MHz. (My choice of AD9361, support 70MHz~6GHz)

ssh root@192.168.10.122
cd openwifi
./wgd.sh
./sdrctl dev sdr0 set reg rf 1 70
./sdrctl dev sdr0 set reg rf 5 70
./fosdem.sh

[image]

[image]

[image]
No peak waveform generation is tested with an oscilloscope. So I'd like to ask what went wrong with my process of setting up 70MHz?
（2）Also I have another question, looking at the logs with dmesg, the power of both AP and STA connected via cable is between -70dBm and -80dBm, is this normal? (Is the power a bit too small, I tried other models of SDR but the power is still the same, how should I fix this?) 
AP：

[image]
STA：

[image]
Looking forward to hearing from you. Thank you.
7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

JiaoXianjun, 2025-03-06:
Can you try running sdrctl again at the end (after fosdem.sh)?

guaguaguaguaguayaya, 2025-03-06:
> Can you try running sdrctl again at the end (after fosdem.sh)?

Thank you very much for your reply. But I don't understand what you mean, maybe you want to ask me about the following situation, if it is not correct, I hope you can explain it to me in more detail.

[image]

guaguaguaguaguayaya, 2025-03-13:
> Can you try running sdrctl again at the end (after fosdem.sh)?

Thank you very much for the hint, I have successfully retrieved the 70MHz signal. Thank you so much!