open-sdr/openwifi #454: How to make openwifi work in 802.11b mode and the phone can successfully connect to the hotspot

jylccc, 2025-04-10:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
yes,i already send it to you
1. Our image is used directly or you build your own image?
use your image
2. What is your own modification?
Try to change the working mode of openwifi to 802.11b.
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
unbuntu18.3 
vivado2018.3 
openwifi branch :master
commit revision：2021.04
4. Board/hardware type
zedboard+fcommcs3
5. WiFi channel number
6
6. Steps to reproduce the issue, and the related error message, screenshot, etc
I want to make openwifi work in 802.11b mode, I refer to the special note for 11b you wrote, I may not understand the meaning, I first changed the support_rate and base_rate in hostapd-openwifi-11ag：

```
interface=sdr0
driver=nl80211
country_code=BE
ssid=openwifi
hw_mode=g
channel=6
#supported_rates=2 4 11 22
#basic_rates=2 4
#supported_rates=60 90 120 180 240 360 480 540
#basic_rates=60 90 120 180
supported_rates=10 20 55 110
basic_rates=10 20 55 110
#supported_rates=60 90
#basic_rates=60 90
#ieee80211n=1
#ht_capab=[SHORT-GI-20]
#require_ht=1
#ieee80211d=1
#ieee80211h=1
#wpa=2
#wpa_passphrase=myrabbit
#wpa_key_mgmt=WPA-PSK
```
After logging into my zedboard with putty, I executed the following command at end point

> cd openwifi
> ./fosedem-11ag.sh 
> ./build_wpa_supplicant_wo11b.sh
./fosdem.sh contains:
```
#!/bin/bash
# Author: Xianjun Jiao
# SPDX-FileCopyrightText: 2019 UGent
# SPDX-License-Identifier: AGPL-3.0-or-later

killall hostapd
killall webfsd

cd ~/openwifi
service network-manager stop
./wgd.sh
ifconfig sdr0 192.168.13.1
route add default gw 192.168.10.1
service isc-dhcp-server restart
hostapd hostapd-openwifi-11ag.conf &
sleep 5
cd webserver
webfsd -F -p 80 -f index.html &
```

build_wpa_supplicant_wo11b.sh contains:

```
#!/bin/bash

# Author: Michael Mehari
# SPDX-FileCopyrightText: 2019 UGent
# SPDX-License-Identifier: AGPL-3.0-or-later

#if [ "$#" -ne 1 ]; then
#    echo "You must enter exactly 1 arguments: \$OPENWIFI_DIR"
#    exit 1
#fi

#OPENWIFI_DIR=$1

set -x

#cd $OPENWIFI_DIR/user_space
wget http://w1.fi/releases/wpa_supplicant-2.1.tar.gz
tar xzvf wpa_supplicant-2.1.tar.gz
patch -d wpa_supplicant-2.1/src/drivers/ < driver_nl80211.patch
cd wpa_supplicant-2.1/wpa_supplicant/
cp defconfig .config
sed -i 's/#CONFIG_LIBNL32.*/CONFIG_LIBNL32=y/g' .config
make clean
make -j16
# sudo make install
cd ../../
rm -r wpa_supplicant-2.1/ wpa_supplicant-2.1.tar.gz

```
The final result is that the wifi signal can be detected on the mobile phone, but when connected, the network refuses the request.
7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

HavingaThijs, 2025-04-14:
Since 802.11b is based on DSSS and not OFDM, you'll have to modify the whole receiver chain running on the FPGA, whose top-level state machine is currently implemented [here](https://github.com/open-sdr/openofdm/blob/39cec9f3ecf3d67069461a3e2bb85d5594712d05/verilog/dot11.v). In principle I believe the MAC layer can remain the same, so indeed you just need to add the decoding of 802.11b, but as this is a whole new PHY (e.g. even packet detection is different) this will require quite some work.

jylccc, 2025-04-15:
Thank you very much for your reply. If I were to implement another 802.11b and get it running successfully, it would be too difficult for one person. Could I hire you or your team to help me complete this work for a fee?

ViruKM, 2025-04-30:
Hi Thijis, 
I am trying build FPGA for Openwifi-hw for antsde200 board. i followed all the steps given in readme. but i am facing lot of errors and i am unable to do ./openwifi.tcl file in vivado and its open openofdm block.

jylccc, 2025-05-19:
> 802.11b support is currently not in our roadmap, for requests regarding business collaboration please contact [Filip.Louagie@UGent.be](mailto:Filip.Louagie@UGent.be).

Thank you very much for your reply. If I want to capture the WiFi received signal level, frequency offset, carrier-to-noise ratio, bit error rate, carrier lock indication, JT lock indication, etc. on the PC side, how can I find these values? Are there clear definitions in the baseband part? The dot11tx.txt file generated by the transmitter and the samples.dat file directly used by the receiver are not the same. The format of the corresponding samples.txt file generated by Python is also different from that of dot11tx.txt. How should I handle it if I want to calculate the bit error rate?

HavingaThijs, 2025-05-19:
You need several different ways to obtain these different values, e.g. 
[this application note](https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi.md) for the frequency offset when capturing real Wi-Fi packets.
Bit error rate is not automatically calculated, you would need to compare the decoded bits with the known bits. 
"Carrier lock" is e.g. the "short preamble detected" counter: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/perf_counter.md

I'm not entirely sure where you are referring to with the `samples.dat` file and `samples.txt` file.

jylccc, 2025-05-20:
> 您需要几种不同的方法来获取这些不同的值，例如[，此应用说明](https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi.md)用于捕获实际 Wi-Fi 数据包时的频率偏移。误码率不是自动计算的，您需要将解码后的位与已知位进行比较。 例如，“Carrier lock” 是 “short preamble detected” 计数器：https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/perf_counter.md
> 
> 我不完全确定你指的是 file 和 file 的什么地方。`samples.dat``samples.txt`

OK, thanks. But where can I find the JT lock and receive level indication? Currently, does OpenWiFi support 2x2 MIMO? If it does, are there any test metrics or test plans for 2x2 MIMO introduced in the MD document?

jylccc, 2025-05-21:
> You need several different ways to obtain these different values, e.g. [this application note](https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi.md) for the frequency offset when [...]

Hello, I would like to ask you a question, that is, I want to make n work at 2.4Ghz, change hw_mode to g,channel set to 6, other things remain the same, but openwifi network does not show up, my code screenshot is as follows.
hostapd-openwifi.conf contains：
`interface=sdr0
driver=nl80211
country_code=BE
ssid=openwifi
hw_mode=g
channel=6
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 90 120 180
ieee80211n=1
#ht_capab=[SHORT-GI-20]
require_ht=1
#ieee80211d=1
#ieee80211h=1
#wpa=2
#wpa_passphrase=myrabbit
#wpa_key_mgmt=WPA-PSK
`
fosdem.sh contains:
`killall hostapd
killall webfsd

cd ~/openwifi
service network-manager stop
./wgd.sh
ifconfig sdr0 192.168.13.1
route add default gw 192.168.10.1
service isc-dhcp-server restart
hostapd hostapd-openwifi.conf &
sleep 5
cd webserver
webfsd -F -p 80 -f index.html &
`

HavingaThijs, 2025-07-12:
It should always leave the `S_WAIT_POWER_TRIGGER` state in simulation, as the `power_thres` and `rssi_half_db` are set to 0 in the testbench, and hence this evaluates to 1: https://github.com/open-sdr/openofdm/blob/39cec9f3ecf3d67069461a3e2bb85d5594712d05/verilog/dot11.v#L272

However, you ran the IQ capture in free running mode, that may mean the capture did not contain any packet. You can run `./side_ch_ctl wh8d9` and `./side_ch_ctl wh11d500` to trigger on short preamble detected and capture 400 IQ samples before that event. See also: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/iq.md#config-the-iq-capture-and-interval 
Note that with the Zedboard you can only capture 4095 IQ samples, so only short packets can be captured completely.

jylccc, 2025-07-12:
> > > I am decoding it in Python using iverilog based on the code in the openofdm repository. Does this affect the speed?
> > 
> > 
> > Decoding in simulation on a CPU will always be slow as it needs to emulate the logic on the FPGA. Decoding more bits in one packet will not help much with this.
> > > If I can log in to the board to control the rate, can I connect my phone to the openwifi network and then capture the actual IQ? When I operate according to the commands in iq.md, I cannot connect to openwifi.
> > 
> > 
> > Yes, follow the instructions to capture the IQ using the application note, but instead of `./monitor_ch.sh sdr0 11` you use `./fosdem.sh` (starting as access point instead of monitoring). Then set the rate with `drv_tx` register 0 and 1.
> > > Could you please tell me how to correctly capture the actual IQ (not the physical layer IQ data)
> > 
> > 
> > I'm not sure what you mean with "actual IQ". The IQ here refers to the physical layer IQ data.
> 
> Sorry, it should be the IQ data of the physical layer. When I tried to connect my Zed development board to capture the IQ data on iq_capture.py and then send it to the openOFDM decoder for decoding, it got stuck in the Power triggered state all the time. Is there something wrong with my operation of capturing the IQ data or is it due to other reasons? Looking forward to your reply.

> It should always leave the `S_WAIT_POWER_TRIGGER` state in simulation, as the `power_thres` and `rssi_half_db` are set to 0 in the testbench, and hence this evaluates to 1: https://github.com/open-sdr/openofdm/blob/39cec9f3ecf3d67069461a3e2bb85d5594712d05/verilog/dot11.v#L272
> 
> However, you ran the IQ capture in free running mode, that may mean the capture did not contain any packet. You can run `./side_ch_ctl wh8d9` and `./side_ch_ctl wh11d500` to trigger on short preamble detected and capture 500 IQ samples before that event. See also: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/iq.md#config-the-iq-capture-and-interval Note that with the Zedboard you can only capture 4095 IQ samples, so only short packets can be captured completely.

I will follow the steps below

cd openwifi
./fosdem-11ag.sh
insmod side_ch.ko iq_len_init=4095
./sdrctl dev sdr0 set reg drv_tx 0 10
./side_ch_ctl wh8d9
./side_ch_ctl wh11d500
./side_ch_ctl  g 

python3 iq_capture.py

result:
Reading memory from...
D:/openofdm/scripts/iq.txt
WARNING: dot11_tb.v:96: $readmemh(D:/openofdm/scripts/iq.txt): Not enough words in the file for the requested range [0:33554431].
Done.
Power triggered.
       100 /           33554432
       200 /           33554432
       300 /           33554432
       400 /           33554432
       500 /           33554432
       600 /           33554432
       700 /           33554432
       800 /           33554432
       900 /           33554432
      1000 /           33554432
can't  deteced the short preamble

HavingaThijs, 2025-07-12:
It looks like you're using the master branch of openOFDM, but openwifi uses this specific commit: https://github.com/open-sdr/openofdm/tree/39cec9f3ecf3d67069461a3e2bb85d5594712d05 
Please try changing to that commit.

jylccc, 2025-07-13:
> It looks like you're using the master branch of openOFDM, but openwifi uses this specific commit: https://github.com/open-sdr/openofdm/tree/39cec9f3ecf3d67069461a3e2bb85d5594712d05 Please try changing to that commit.

I tried to compile the Verilog code in openofdm of openwifi with iverilog, but got the following error:
mv_avg.v:56: error: Unknown module type: xpm_fifo_sync
fifo_sample_delay.v:53: error: Unknown module type: xpm_fifo_sync
mv_avg_dual_ch.v:66: error: Unknown module type: xpm_fifo_sync
mv_avg_dual_ch.v:66: error: Unknown module type: xpm_fifo_sync
But I can't find this module anywhere
//`include "openofdm_rx_pre_def.v"，I have annotated this line, but if I don't annotate it, an error will still occur. It seems that I cannot find this module

HavingaThijs, 2025-07-13:
You'll need Xilinx tools to simulate it, please follow the instructions of openwifi-hw: https://github.com/open-sdr/openwifi-hw?tab=readme-ov-file#simulate-ip-cores