open-sdr/openwifi #457: openwifi porting problem.

18959263172, 2025-04-17:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
already sent.
1. Our image is used directly or you build your own image?
builit my own image based on 1.4 and sdrpi hdl-hw project.
2. What is your own modification?
change hdl xdc.for my custome board. with vivado 2022 tool.
3. Versions: OS, Vivado, openwifi/openwifi-hw 
UBUNUTU,2022.2. sdrpi branch
4. Board/hardware type
ad9363/zynq020 designed by myself.
5. WiFi channel number
no reach this step
6. Steps to reproduce the issue, and the related error message, screenshot, etc

I want to port openwifi to my own board.
I use sdrpi branch.
And I can successful run openwifi.  (wgd.sh.foden.sh),it can detect sdr0. but my phone cannot detect wifi.
I'm pretty sure my xdc file is ok because the config work well in plutosdr fw.

here is the infomation in linux.

[image]

[image]

I add ILA to monitor adc data for openwifi, looks like the adc clock is wrong

[image]

[image]

It is strange because I don't see where this clock come from.

[image]

the ila is driven by fclk3 of zynq ,250MHZ.

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

dmesg result:

[    0.242169] zynq-pinctrl 700.pinctrl: zynq pinctrl initialized
[    0.247292] e0001000.serial: ttyPS0 at MMIO 0xe0001000 (irq = 25, base_baud = 6249999) is a xuartps
[    0.255233] console [ttyPS0] enabled
[    0.259675] bootconsole [earlycon0] disabled
[    0.323296] SCSI subsystem initialized
[    0.325929] usbcore: registered new interface driver usbfs
[    0.330147] usbcore: registered new interface driver hub
[    0.335750] usbcore: registered new device driver usb
[    0.340085] media: Linux media interface: v0.10
[    0.343669] Linux video capture interface: v2.00
[    0.347194] pps_core: LinuxPPS API ver. 1 registered
[    0.350850] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.358892] PTP clock support registered
[    0.362326] FPGA manager framework
[    0.364932] Advanced Linux Sound Architecture Driver Initialized.
[    0.370861] clocksource: Switched to clocksource arm_global_timer
[    0.390640] NET: Registered protocol family 2
[    0.394859] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[    0.400638] TCP bind hash table entries: 4096 (order: 3, 32768 bytes)
[    0.406365] TCP: Hash tables configured (established 4096 bind 4096)
[    0.411650] UDP hash table entries: 256 (order: 1, 8192 bytes)
[    0.416194] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes)
[    0.421366] NET: Registered protocol family 1
[    0.424803] hw perfevents: no interrupt-affinity property for /pmu@f8891000, guessing.
[    0.431609] hw perfevents: enabled with armv7_cortex_a9 PMU driver, 7 counters available
[    0.441045] workingset: timestamp_bits=30 max_order=17 bucket_order=0
[    0.449886] io scheduler noop registered
[    0.452550] io scheduler deadline registered
[    0.455550] io scheduler cfq registered (default)
[... 368 lines cut ...]
[ 1021.540757] sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_TXPOWER 14
[ 1021.540766] sdr,sdr openwifi_config changed flag 00000002
[ 1021.540780] sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_BEACON_INT 64
[ 1021.540785] sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_TXPOWER 14
[ 1021.540789] sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_BEACON_ENABLED
[ 1021.543393] sdr,sdr openwifi_bss_info_changed WARNING enable_beacon
[ 1021.543404] sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_BEACON_ENABLED 512 BSS_CHANGED_BEACON 256
[ 1021.543471] IPv6: ADDRCONF(NETDEV_CHANGE): sdr0: link becomes ready
[ 1021.543663] sdr,sdr openwifi_prepare_multicast
[ 1021.543681] sdr,sdr openwifi_configure_filter MON 0 M_BCN 1 BST0 1 BST1 1 UST 1 PB_RQ 1 PS_PL 1 O_BSS 0 CTL 0 BCN_PRP 0 PCP_FL 0 FCS_FL 0 ALL_MUT 1
[ 1021.543921] sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_ERP_CTS_PROT 0
[ 1021.543928] sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_BASIC_RATES f0
[ 1021.543934] sdr,sdr openwifi_bss_info_changed WARNING BSS_CHANGED_ERP_SLOT 8 BSS_CHANGED_ERP_PREAMBLE 4 short slot 1
[ 1021.544985] sdr,sdr openwifi_conf_tx: [queue 0], aifs: 1, cw_min: 3, cw_max: 7, txop: 47, aifs and txop ignored
[ 1021.545055] sdr,sdr openwifi_conf_tx: [queue 1], aifs: 1, cw_min: 7, cw_max: 15, txop: 94, aifs and txop ignored
[ 1021.545118] sdr,sdr openwifi_conf_tx: [queue 2], aifs: 3, cw_min: 15, cw_max: 63, txop: 0, aifs and txop ignored
[ 1021.545180] sdr,sdr openwifi_conf_tx: [queue 3], aifs: 7, cw_min: 15, cw_max: 1023, txop: 0, aifs and txop ignored
[ 1023.564042] sdr,sdr openwifi_prepare_multicast
[ 1023.564063] sdr,sdr openwifi_configure_filter MON 0 M_BCN 1 BST0 1 BST1 1 UST 1 PB_RQ 1 PS_PL 1 O_BSS 0 CTL 0 BCN_PRP 0 PCP_FL 0 FCS_FL 0 ALL_MUT 1
root@analog:~/openwifi# 

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better? 
please any suggestion.

ViruKM, 2025-04-29:
Hi 
i am also working on openwifi-hw build fpga design, but i am unable to open ./openwifi.tcl file after am getting lot of errors.

18959263172, 2025-05-10:
I'm tring to debug it. error:

root@analog:~/openwifi# ./inject_80211/inject_80211 -m n -r 0 -n 10 -s 64 sdr0
DLT_IEEE802_11_RADIO Encap
mode = 802.11n, rate index = 0, SHORT G8<--- cut here ---
I = 0, number of packets = 10 and packet size = 120 bytes, delayUnable to handle kernel NULL pointer dereference at virtual address 0000001a
 = 100000 usec
packet_type d sub_type 1 payload_len 68 ieee_hdrpgd = 6f443a3f
_len 24 addr1 01 addr2 02
[0000001a] *pgd=07d2f831, *pte=00000000, *ppte=00000000
Internal error: Oops: 17 [#1] PREEMPT SMP ARM
Modules linked in: sdr(O) xpu(O) openofdm_rx(O) openofdm_tx(O) rx_intf(O) tx_intf(O) mac80211 libarc4 xilinx_dma ad9361_drv cfg80211 8021q mwipcore mwipcore_iio_streaming mwipcore_dma_streaming mathworks_ip_common ipv6
CPU: 0 PID: 5150 Comm: inject_80211 Tainted: G           O      5.10.0-19493-g1bbe32fa5182-dirty #5
Hardware name: Xilinx Zynq Platform
PC is at openwifi_tx+0xe0c/0x1468 [sdr]
LR is at openwifi_tx+0x354/0x1468 [sdr]
pc : [<bf1b61f4>]    lr : [<bf1b573c>]    psr: 60000013
sp : c7cdfb30  ip : 00000000  fp : 00000000
r10: c2c266c0  r9 : c2d0085e  r8 : 00000008
r7 : ce5fd7e0  r6 : c2c266c0  r5 : ce5fd7c0  r4 : 00000000
r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : 00000001
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 18c5387d  Table: 06bd004a  DAC: 00000051
Process inject_80211 (pid: 5150, stack limit = 0x9eb2a3d5)
Stack: (0xc7cdfb30 to 0xc7ce0000)
fb20:                                     00000000 1ec44000 c7cdfb4c c01448b0
fb40: dfb93a00 c1454080 c7cdfb64 c01448cc c1454080 00000000 dfb93a00 c1454514
fb60: c7cdfb94 c0145044 00000001 2538b031 0000e666 2538b031 c1463f6c a0000113
fb80: 00000000 0000005c 00000000 00000002 00000002 00000218 00000000 ce5fea04
fba0: ce5fc500 00000000 0000005c 00000000 00000002 00000000 00000001 00000000
fbc0: c7cdfc58 00000008 ce5fc9e4 bf110400 01223344 00000000 00000001 00000000
fbe0: 00000000 00000000 dfb91f30 00005566 c0f49340 01223344 00005566 000000d9
fc00: 02223344 00005566 00000000 00000000 00060003 00180020 c2c266c0 2538b031
fc20: 33445566 c7cdfcb4 ce5fc500 c2c266c0 c7cdfcb4 ce5fc910 bf1840cd bf179fd4
fc40: 00000002 bf135bb0 00000000 00000000 00000000 00000000 00000000 2538b031
fc60: ce5fc500 00000000 c7cdfcb4 ce5fc500 00000008 0000005c ce5fc500 00000078
fc80: 00000218 bf135ccc 00000000 c805e540 c2c266c0 00000000 c805e540 00000000
fca0: 0000005c bf13c764 00000000 ce5fc1c0 00000000 c7cdfcb4 c7cdfcb4 00000000
fcc0: 00000000 ce5fc500 c805e540 00000000 00000000 00000000 00000002 2538b031
fce0: c2c266c0 c2c266c0 c805e540 ce5fc1c0 0000001c ce5fd4f8 c2d00842 bf13ca08
fd00: bf13c88c c2c266c0 c805e000 00000000 c2224600 c1002d00 c0f5033a c1006350
fd20: c0f4e86c c0766598 00000000 c7cdfd60 c805e000 c2ef4c00 c2c266c0 c2224600
fd40: 00000000 00000000 c805e000 c2ef4c90 c2c266c0 c079ca38 00000001 0000001c
fd60: 00000010 2538b031 00000000 c2ef4c00 00000001 c2c266c0 00000040 00000001
fd80: c2ef4c40 c079cd0c 00000000 00000001 0000000e bf121e58 c805e000 00000000
fda0: c805e000 c2ef4c00 c2224600 00000000 c2d00900 c2d00900 0000000e c07669e8
fdc0: c7d1b800 c2c266c0 c805e000 fffffff4 00000000 c08a816c 00000000 00000000
fde0: 00000000 2538b031 c7d1b800 00000078 c289e400 c7d1b800 c2c266c0 c805e000
fe00: c2d00900 c08ab228 c7cdfe60 00000000 00000000 c7cdfec8 00000300 00000000
fe20: 00000000 00000050 00000078 00000000 c1549040 c041d6f8 c1549040 00000000
fe40: 00000000 00000000 c18025d8 c1802628 c0db1ee8 00080040 00080060 c18025d8
fe60: 00000000 00000000 00000000 c1800000 00000000 00000000 00000000 00000000
fe80: 00000000 2538b031 00000000 00000000 c289e400 00000040 00000000 c0100264
fea0: c7cde000 00000121 bec07a94 c0743418 c7cdfed0 c7cde000 fffffff7 00000000
fec0: bec0744c 00000078 00000000 00000000 00000005 00000000 00000000 c7cdfec8
fee0: 00000000 0000004a 00000000 c7cdff80 00000000 00000040 bec06d3c c02211c0
ff00: b6e0f000 c6bd2db8 c6bd2db8 00000000 00000000 00000000 00000000 dfcb0e14
ff20: 00000000 2538b031 00000000 c7cdffb0 b6e0fd20 2538b031 c6b4ac00 c7cdffb0
ff40: b6e0fd20 80000007 c6b4ac00 c011150c 00000004 00000004 c6a5ecc0 2538b031
ff60: 1b6b02b7 c8340cc0 c8340cc0 00000000 00000000 c0100264 c7cde000 2538b031
ff80: 00000000 00000004 b6fc6500 000107f8 00000121 c0743490 00000000 00000000
ffa0: 00000004 c0100060 00000004 b6fc6500 00000004 bec0744c 00000078 00000000
ffc0: 00000004 b6fc6500 000107f8 00000121 00000000 00000000 b6fce000 bec07a94
ffe0: b6ea74d0 bec06d38 b6ea74dc b6ea74f8 60000010 00000004 00000000 00000000
[<bf1b61f4>] (openwifi_tx [sdr]) from [<bf135bb0>] (ieee80211_tx_frags+0x228/0x2c4 [mac80211])
[<bf135bb0>] (ieee80211_tx_frags [mac80211]) from [<bf135ccc>] (__ieee80211_tx+0x80/0x194 [mac80211])
[<bf135ccc>] (__ieee80211_tx [mac80211]) from [<bf13c764>] (ieee80211_tx+0x110/0x140 [mac80211])
[<bf13c764>] (ieee80211_tx [mac80211]) from [<bf13ca08>] (ieee80211_monitor_start_xmit+0x17c/0x240 [mac80211])
[<bf13ca08>] (ieee80211_monitor_start_xmit [mac80211]) from [<c0766598>] (dev_hard_start_xmit+0xc8/0x13c)
[<c0766598>] (dev_hard_start_xmit) from [<c079ca38>] (sch_direct_xmit+0xf4/0x2b0)
[<c079ca38>] (sch_direct_xmit) from [<c079cd0c>] (__qdisc_run+0x118/0x520)
[<c079cd0c>] (__qdisc_run) from [<c07669e8>] (__dev_queue_xmit+0x310/0x754)
[<c07669e8>] (__dev_queue_xmit) from [<c08ab228>] (packet_sendmsg+0xa78/0x12fc)
[<c08ab228>] (packet_sendmsg) from [<c0743418>] (__sys_sendto+0xd0/0x12c)
[<c0743418>] (__sys_sendto) from [<c0743490>] (sys_send+0x18/0x20)
[<c0743490>] (sys_send) from [<c0100060>] (ret_fast_syscall+0x0/0x54)
Exception stack(0xc7cdffa8 to 0xc7cdfff0)
ffa0:                   00000004 b6fc6500 00000004 bec0744c 00000078 00000000
ffc0: 00000004 b6fc6500 000107f8 00000121 00000000 00000000 b6fce000 bec07a94
ffe0: b6ea74d0 bec06d38 b6ea74dc b6ea74f8
Code: 13a03001 e58d3058 e59d3090 e5933000 (e1d321ba) 
---[ end trace 3db150cf1ee880b3 ]---
Kernel panic - not syncing: Fatal exception in interrupt
---[ end Kernel panic - not syncing: Fatal exception in in